MLB-LED-Scoreboard/mlb-led-scoreboard #535: Error Message when trying to start the scoreboard

PHI34Halladay, 2024-04-08:
## This issue is a

- [x] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): 4B
- Operating System: raspberry
- Matrix size: 128x64
- Display adapter (HAT/Bonnet/etc.): adafruit-hat-pwm
  - Additional information:
- Power source(s):

## Software Configuration

- MLB LED Scoreboard Version: 8.0.1
- Commands/arguments used to start the scoreboard: sudo python3 /home/pi/mlb-led-scoreboard/main.py --led-rows=32 --led-cols=64 --led-chain=4 --led-gpio-mapping=adafruit-hat-pwm --led-pixel-mapper=U-mapper --led-brightness=30

## Expected behavior
Scoreboard should start

## Actual behavior
Scoreboard doesn't start

## Additional Information
Guys, I'm running the MLB Scoreboard on my raspberry. I normally start the scoreboard by jumping in the mlb-led-scoreboard folder through cd mlb-led-scoreboard and then I run the command from above which works as it should.

I got myself a Homeassistant and I'm trying to create an automation that connects both the HA and my scoreboard.

All I need for this is just one command. So I tried using this command here:

sudo python3 /home/pi/mlb-led-scoreboard/main.py --led-rows=32 --led-cols=64 --led-chain=4 --led-gpio-mapping=adafruit-hat-pwm --led-pixel-mapper=U-mapper --led-brightness=30

I get this message in return:

pi@raspberrypi:~ $ sudo python3 /home/pi/mlb-led-scoreboard/main.py --led-rows=32 --led-cols=64 --led-chain=4 --led-gpio-mapping=adafruit-hat-pwm --led-pixel-mapper=U-mapper --led-brightness=30
ERROR (04:04:23): Untrapped error in main!
Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 169, in <module>
    main(matrix, config)
  File "/home/pi/mlb-led-scoreboard/main.py", line 41, in main
    config = Config(config_base, matrix.width, matrix.height)
  File "/home/pi/mlb-led-scoreboard/data/config/__init__.py", line 26, in __init__
    self.preferred_teams = json["preferred"]["teams"]
KeyError: 'preferred'

How would the correct command be?

Charliethom, 2024-04-10:
Don"t know if this will help or not, I am running s similar setup. First off I start mine from the mlb-led-scoreboard directory. My start command is "sudo python3 main.py  --led-gpio-mapping=adafruit-hat-pwm --led-rows=64 --led-cols=64 --led-chain=2 --led-brightness=80 --led-slowdown-gpio=2 --led-no-hardware-pulse=2"
I am currently running 3 similar boards with same commands other than matrix size and rows and cols.
Hope this helps.

ty-porter, 2024-04-10:
The command to start the scoreboard has changed in v8 compared to v7 and under.

https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/blob/master/README.md#usage

```sh
sudo ./main.py <FLAGS>
```

PHI34Halladay, 2024-04-10:
Hi @ty-porter, that might be the missing part in my project :)

I will try it as soon as I get home.

Thanks in advance

Charliethom, 2024-04-10:
Thanks for the tip, that worked.

PHI34Halladay, 2024-04-11:
Don't think that this command will work tho:

sudo python3 /home/pi/mlb-led-scoreboard/./main.py --led-rows=32 --led-cols=64 --led-chain=4 --led-gpio-mapping=adafruit-hat-pwm --led-pixel-mapper=U-mapper --led-brightness=30

I need one command that starts the scoreboard. 

Right now I'm doing two commands by 1) going into the right folder (cd mlb-led-scoreboard) and 2) start it (sudo python3 ./main.py ...)
That needs to be put into just one command.

ty-porter, 2024-04-11:
> Don't think that this command will work tho:
> 
> sudo python3 /home/pi/mlb-led-scoreboard/./main.py --led-rows=32 --led-cols=64 --led-chain=4 --led-gpio-mapping=adafruit-hat-pwm --led-pixel-mapper=U-mapper --led-brightness=30

It will not, you need to use the command I showed above. You do not run this via `python3` now. The script itself is executable and locates its own Python.

```
sudo ./home/pi/mlb-led-scoreboard/main.py <flags>
```

> Right now I'm doing two commands by 1) going into the right folder (cd mlb-led-scoreboard) and 2) start it (sudo python3 ./main.py ...) That needs to be put into just one command.

Can say with confidence that it does not.

tracerrx, 2025-03-09:
Closing this issue.  If the problem persists please create a new issue

closed 2025-03-09 by tracerrx