MLB-LED-Scoreboard/mlb-led-scoreboard #536: Button to turn on/off scoreboard

Charliethom, 2024-04-21:
## Expected behavior

## Actual behavior

## Additional Information
I installed a momentary button on my adafruit hat on a previous scoreboard setup I have and it would turn the scoreboard on/off. I know the pins that are used but I can not for the life of me find the script used to do this for my pi4 setup. Can anyone help with this?
Thanks in advance.

ty-porter, 2024-04-21:
This is not a supported feature for this project and doesn't really belong in the issue tracker.

There are many examples of what you are looking to do elsewhere. For instance, here: https://raspberrypi.stackexchange.com/questions/117013/raspberry-pi-4-b-gpio-boot-and-shutdown-buttons

We do welcome these topics in the discussion section or on Discord.

closed 2024-04-21 by ty-porter

unwound29, 2024-04-21:
Lower tech, might be easier to switch off the display by using a power cord like [this?](https://www.amazon.ca/Enokay-Raspberry-Supply-Charger-Adapter/dp/B07VT554J1/ref=sr_1_9)