MLB-LED-Scoreboard/mlb-led-scoreboard #537: No matching distribution found for Pillow>=10.0.1

ChickenSupreme, 2024-04-22:
## This issue is a

- [x] Bug Report
- [ ] Feature Request

## Hardware Configuration

- Raspberry Pi Revision (4B+/3B+/Zero/etc.): Raspberry Pi 4 Model B Rev 1.4
- Operating System: Raspbian GNU/Linux 10 (buster)
- Matrix size: 64x32
- Display adapter (HAT/Bonnet/etc.): Adafruit
  - Additional information:
- Power source(s):

## Software Configuration

- MLB LED Scoreboard Version: MLB LED Scoreboard v8.0.1
- Commands/arguments used to start the scoreboard:

```
[Unit]
Description=MLB LED Scoreboard
Wants=network-online.target time-sync.target
After=network.target network-online.target time-sync.target

[Service]
Environment="SCOREBOARD_ARGS=--led-gpio-mapping=\"adafruit-hat-pwm\" --led-slowdown-gpio=4 --led-rows=32 --led-cols=64 --led-brightness=60"
WorkingDirectory=/home/pi/mlb-led-scoreboard
ExecStart=/usr/bin/python3 main.py $SCOREBOARD_ARGS
Restart=always
RestartSec=45

[Install]
WantedBy=multi-user.target
```

## Expected behavior

Run install.sh and everything installs properly.

## Actual behavior

Run install.sh and Pillow fails to install.

```
sudo ./install.sh 

------------------------------------
  Installing python 3...
------------------------------------

Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'python3-pil' instead of 'python3-pillow'
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
libopenjp2-7 is already the newest version (2.3.0-2+deb10u2).
libsdl2-image-2.0-0 is already the newest version (2.0.4+dfsg1-1+deb10u1).
libsdl2-mixer-2.0-0 is already the newest version (2.0.4+dfsg1-1).
libsdl2-ttf-2.0-0 is already the newest version (2.0.15+dfsg1-1).
libxml2-dev is already the newest version (2.9.4+dfsg1-7+deb10u6).
libxslt1-dev is already the newest version (1.1.32-2.2~deb10u2).
python3-dev is already the newest version (3.7.3-1).
python3-pil is already the newest version (5.4.1-2+deb10u6).
python3-tk is already the newest version (3.7.3-1).
python3-venv is already the newest version (3.7.3-1).
libsdl2-2.0-0 is already the newest version (2.0.9+dfsg1-1+rpt1).
python3-pip is already the newest version (18.1-5+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

------------------------------------
  Installing dependencies...
------------------------------------

Creating virtual environment...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: feedparser==6.0.10 in ./venv/lib/python3.7/site-packages (from -r requirements.txt (line 1)) (6.0.10)
Requirement already satisfied: MLB_StatsAPI>=1.6.1 in ./venv/lib/python3.7/site-packages (from -r requirements.txt (line 2)) (1.7.1)
Collecting Pillow>=10.0.1 (from -r requirements.txt (line 3))
  Could not find a version that satisfies the requirement Pillow>=10.0.1 (from -r requirements.txt (line 3)) (from versions: 1.0, 1.1, 1.2, 1.3, 1.4, 1.5, 1.6, 1.7.0, 1.7.1, 1.7.2, 1.7.3, 1.7.4, 1.7.5, 1.7.6, 1.7.7, 1.7.8, 2.0.0, 2.1.0, 2.2.0, 2.2.1, 2.2.2, 2.3.0, 2.3.1, 2.3.2, 2.4.0, 2.5.0, 2.5.1, 2.5.2, 2.5.3, 2.6.0, 2.6.1, 2.6.2, 2.7.0, 2.8.0, 2.8.1, 2.8.2, 2.9.0, 3.0.0, 3.1.0rc1, 3.1.0, 3.1.1, 3.1.2, 3.2.0, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.4.0, 3.4.1, 3.4.2, 4.0.0, 4.1.0, 4.1.1, 4.2.0, 4.2.1, 4.3.0, 5.0.0, 5.1.0, 5.2.0, 5.3.0, 5.4.0, 5.4.1, 6.0.0, 6.1.0, 6.2.0, 6.2.1, 6.2.2, 7.0.0, 7.1.0, 7.1.1, 7.1.2, 7.2.0, 8.0.0, 8.0.1, 8.1.0, 8.1.1, 8.1.2, 8.2.0, 8.3.0, 8.3.1, 8.3.2, 8.4.0, 9.0.0, 9.0.1, 9.1.0, 9.1.1, 9.2.0, 9.3.0, 9.4.0, 9.5.0)
No matching distribution found for Pillow>=10.0.1 (from -r requirements.txt (line 3))
Running rgbmatrix installation...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
make is already the newest version (4.2.1-1.2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
mkdir: cannot create directory ‘submodules’: File exists
fatal: destination path 'matrix' already exists and is not an empty directory.
Already up to date.
make -C ./lib
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[2]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[2]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
/home/pi/mlb-led-scoreboard/venv/bin/python3 setup.py build --build-lib .
running build
running build_py
running build_ext
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/bindings/python'
make -C ./lib
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[2]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
make[2]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/lib'
/home/pi/mlb-led-scoreboard/venv/bin/python3 setup.py build --build-lib .
running build
running build_py
running build_ext
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/bindings/python'
make -C bindings/python install
make[1]: Entering directory '/home/pi/mlb-led-scoreboard/submodules/matrix/bindings/python'
/home/pi/mlb-led-scoreboard/venv/bin/python3 setup.py install
running install
running build
running build_py
running build_ext
running install_lib
running install_egg_info
Removing /home/pi/mlb-led-scoreboard/venv/lib/python3.7/site-packages/rgbmatrix-0.0.1.egg-info
Writing /home/pi/mlb-led-scoreboard/venv/lib/python3.7/site-packages/rgbmatrix-0.0.1.egg-info
make[1]: Leaving directory '/home/pi/mlb-led-scoreboard/submodules/matrix/bindings/python'
------------------------------------
  Checking for snd_bcm2835
------------------------------------
Sound Blacklist File found, skipping creation.

===================================================================================
  If you have custom configurations, colors, or coordinates, it's recommended to
  update them with the latest options at this time.

  This operation is automatic and will ensure you have up-to-date configuration.

  This action will NOT override any custom configuration you already have unless
  the option has been obsoleted and is no longer in use.
===================================================================================

Would you like to do this now? [Y/n] y

Fetching custom config files...
  - Found custom configuration at ./config.json!
      Configuration is up-to-date.

Installation finished!
```

## Additional Information

This pi was running mlb-led-scoreboard properly for some time but sat powered off for a year. It was plugged in again a few weeks ago and the system updates were applied and it was still working properly. I just ran a git pull then install.sh and now it's broken and fails to run. It appears the install script is unable to find a working version of Pillow. The error message from systemd is:

`ERROR (19:26:28): We require MLB-StatsAPI 1.5.1 or higher. You may need to re-run install.sh`

pjockey, 2024-04-22:
Which version of Raspbian and Python are you running? Version 8 is a bit of a major overhaul and requires everything in the virtual environment (venv) which the install script does for you.

ChickenSupreme, 2024-04-23:
I'm honestly not sure. I realized after posting this that it would be easier to just reformat the SD card and start again from scratch. I'm in the middle of that now.

It was probably just an old OS and/or Python version. Sorry for the bother, closing this now.

closed 2024-04-23 by ChickenSupreme

ty-porter, 2024-04-23:
I agree with reflashing your OS -- this looks like it was installing for Python 3.7 which we no longer support.

We require at least Python 3.11 now, which should be the system Python on all latest Raspberry Pi OS versions.

Just be careful to follow the new startup command syntax as that has changed considerably from the one you posted. Details are in the README.

ChickenSupreme, 2024-04-23:
Ok yeah I was able to get it running again pretty easily with a fresh install. Thanks for the quick replies!