MLB-LED-Scoreboard/mlb-led-scoreboard #545: Display is incorrect when using qty2 64x32 Waveshare displays

ando01, 2024-06-30:
I have the following hardware:
RPi4
Adafruit Bonnet (with E pin soldered to 8)
5V 10A Power supply
2 WaveShare 64x32 LED Matrix Displays

Here are pictures of the hardware connections:

[image]

[image]

Here is the command line I am using:
sudo ./main.py --led-gpio-mapping="adafruit-hat" --led-slowdown-gpio=3 --led-rows 64 --led-cols 64 --led-chain 2

Here is what my displays look like after running that command:

[image]

I cannot figure out why this is not working and hoping someone here can help.

Thanks in advance

ty-porter, 2025-03-09:
`--led-parallel 2` is likely what you need

`--led-chain` is for horizontal chains, which would be 128x32 here

What you want is 2 chains of 1 panel each I think

ando01, 2025-03-13:
Thanks @ty-porter I did fix this a while ago and forgot to close.

closed 2025-03-13 by ando01