MLB-LED-Scoreboard/mlb-led-scoreboard #549: MLBStats and RGB-Matrix not installing correctly.

mcangeli, 2024-08-01:
## This issue is a

- [x ] Bug Report
- [ ] Feature Request

## Hardware Configuration

Raspi 4B 2gb
Adafruit RGB Matrix Bonnet
64x32 RGB LED Matrix
(The suggested BOL)

## Software Configuration
Raspberrypi Bookworm fresh install
MLB Scoreboard fresh install

- MLB LED Scoreboard Version:
- Commands/arguments used to start the scoreboard:

On a fresh install it appears that the MLB StatsAPI and the RGB Matrix are not installed correctly. I was able to get the stats installed, however I had to install the RGB Matrix with the install script from adafruit (which works really well).

However, now I have to use: sudo python main.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat

to start the board.

lambtor, 2024-08-02:
did you install the rgb matrix from within the mlb-led-scoreboard directory?  i'm having the same problem.  statsAPI did not install on 32 bit, but seems to have installed ok on 64 bit.

mcangeli, 2024-08-05:
> did you install the rgb matrix from within the mlb-led-scoreboard directory? i'm having the same problem. statsAPI did not install on 32 bit, but seems to have installed ok on 64 bit.

I ended up running 
" pip3 install pipreqs --break-system-packages"

in the MLB-Scoreboard directory and it installed the statsapi.

I did have to install the rgbmatrix from a different directory (as the install for the mlb-scoreboard already tried to install it).
I ran the install script that adafruit has and it installed ok.

lambtor, 2024-08-12:
do you have the command you used to install the rgb matrix adafruit script?

mcangeli, 2024-08-12:
`curl https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/main/rgb-matrix.sh >rgb-matrix.sh
sudo bash rgb-matrix.sh`

craigw97, 2024-08-14:
I'm having issues following the steps above, but it seems to mostly be due to permissions issues on the rpi4. After I do a clean download of the Mlb-led-scoreboard when I try to do 'pip3 install pipreqs --break-system-packages' I get an ESError: [Errno 13] error stating that the permission was denied. Is the rpi4 misconfigured or should I be doing all of this in a virtual environment? Thanks for the help.

pjockey, 2024-08-14:
Did you try it sudo in front of the command?

craigw97, 2024-08-14:
Putting Sudo in front of the command got it to work, however, I'm still having issues getting the script to work. I downloaded the rbg-matrix via the code script Mcangeli shared. I verified it worked by navigating to mlb-led-scoreboard/submodules/matrix/examples-api-use and running the demo code. With this, the LED screen works perfectly on the examples. When I navigate back to the mlb-led-scoreboard folder and run:
`sudo ./main.py --led-rows=32 --led-cols=64 --led-no-hardware-pulse=1 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4`
I'm getting the errors stating that the SixelAdapter failed to load, and it later states that ModuleNotFoundError: No module named 'rgbmatrix'
Do I need to be running this in a virtual environment or am I missing something to get this to work?

pjockey, 2024-08-14:
If everything was installed correctly, the command should take you into the venv folder and run everything from inside there.
[image]

This how my file system looks on the Pi.

I don't see anything in the instructions about `pipreqs`. There is no file on my Pi. The `sudo ./install.sh` should locate and install all of the required files/packages from the `requirements.txt` file.

Contents of requirements.tx:feedparser==6.0.10
```
feedparser==6.0.10
MLB_StatsAPI>=1.6.1
Pillow>=10.0.1
pyowm==3.3.0
RGBMatrixEmulator>=0.8.4
tzlocal==4.2
```
```
Requirements
You need Git for cloning this repo and PIP for installing the scoreboard software.

sudo apt-get update
sudo apt-get install git python3-pip

Installing the scoreboard software
This installation process will take about 10-15 minutes. Raspberry Pis aren't the fastest of computers, so be patient!

git clone https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard.git
cd mlb-led-scoreboard/
sudo ./install.sh

This will create a Python Virtual Environment and install all of the required dependencies. The virtual environment will be located at mlb-led-scoreboard/venv/.
```

craigw97, 2024-08-15:
Thank you for sharing that. Looking at my file system, I'm seeing the below files. 
`x@raspberrypi:~/Final2/mlb-led-scoreboard $ ls
assets                highlight.py              requirements.txt
colors                install.sh                rgb-matrix.sh
config.json           ISSUE_TEMPLATE.md         submodules
config.json.example   LICENSE.md                tests
CONTRIBUTING.md       main.py                   utils.py
coordinates           PULL_REQUEST_TEMPLATE.md  validate_config.py
data                  __pycache__               venv
debug.py              pyproject.toml            version.py
driver                README.md
emulator_config.json  renderers
`
It looks like I'm missing some folders, like bin, lib, share, include and the pyvenv.cfg file. My file system also seems to include rgb-matrix.sh. Is it possible I installed the rgb-matrix into the wrong location?
Also, my requirements.txt seem to be slightly different. Is this just a difference in version of the software? Thank you for the help, I realise this is fully operator error on my end.
`core==1.0.1
feedparser==6.0.10
Pillow==9.4.0
Pillow==10.4.0
pyowm==3.3.0
RGBMatrixEmulator==0.11.6
tzlocal==4.2`

longbored, 2024-09-10:
Unfortunately, I'm experiencing this same issue, and have followed all of the tips in here exhaustively. I even was able to install rgbmatrix from the local files in the submodules. If I run Python either in the venv, or out, and simply import rgbmatrix, it imports without issues.

But once I actually go to run main.py (again, either in the venv, or outside of it), it can't figure things out. Sadly, I'm very stuck.

tracerrx, 2025-03-09:
Closing this issue.  If the problem persists please create a new issue.

closed 2025-03-09 by tracerrx