MLB-LED-Scoreboard/mlb-led-scoreboard #565: Noob here. Can’t figure out how to set up for my panel size.

Roach097, 2025-01-15:
My 64 x 32 panel is duplicating itself. I’m not quite sure what I’m doing wrong. I’ve messed with the 64 x 32 example json file but nothing has seemed to change. I was able to edit and choose my favorite team and standings. I just need to figure out why this is duplicating..
[image]

c2mfj, 2025-01-15:
Are you adding --led-cols = 64 to your arguments?

Roach097, 2025-01-15:
Well, I just did what you suggested, and that was somewhat of an improvement, but still glitching back-and-forth. [image][image]

Roach097, 2025-01-16:
Update, it was my auto launch script. I had to add the rows and columns dimensions into it

closed 2025-01-19 by Roach097