MLB-LED-Scoreboard/mlb-led-scoreboard #568: sudo ./main.py --emulated throws NUMPY error

Iainrach, 2025-02-09:
Just love this Scoreboard so much - I though I would set up another one as I have a working version on a Libre Le Potato ....

However not as easy as I hoped..................

Pi4 4GB with fresh  - Raspberry Pi OS with desktop and recommended software

Full installation :- 

git clone https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard.git
cd mlb-led-scoreboard/
sudo ./install.sh

Then : -

sudo ./main.py --emulated

I get this error

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/numpy/_core/__init__.py", line 23, in <module>
    from . import multiarray
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/numpy/_core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/numpy/_core/overrides.py", line 7, in <module>
    from numpy._core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/numpy/__init__.py", line 114, in <module>
    from numpy.__config__ import show_config
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy._core._multiarray_umath import (
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/numpy/_core/__init__.py", line 49, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/home/pi/mlb-led-scoreboard/venv/bin/python3"
  * The NumPy version is: "2.2.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/./main.py", line 28, in <module>
    import driver
  File "/home/pi/mlb-led-scoreboard/driver/__init__.py", line 50, in <module>
    sys.modules['driver'] = DriverWrapper()
                            ^^^^^^^^^^^^^^^
  File "/home/pi/mlb-led-scoreboard/driver/__init__.py", line 12, in __init__
    self.set_mode(DriverMode.SOFTWARE_EMULATION)
  File "/home/pi/mlb-led-scoreboard/driver/__init__.py", line 42, in set_mode
    import RGBMatrixEmulator
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/RGBMatrixEmulator/__init__.py", line 2, in <module>
    from RGBMatrixEmulator.emulation.matrix import RGBMatrix
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/RGBMatrixEmulator/emulation/matrix.py", line 1, in <module>
    from RGBMatrixEmulator.emulation.canvas import Canvas
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/RGBMatrixEmulator/emulation/canvas.py", line 1, in <module>
    import numpy as np
  File "/home/pi/mlb-led-scoreboard/venv/lib/python3.11/site-packages/numpy/__init__.py", line 119, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

WardBrian, 2025-02-10:
Can you try again after running `sudo apt install libopenblas-base libopenblas-dev`

Iainrach, 2025-02-10:
Thx - That Worked !

closed 2025-02-10 by WardBrian