MLB-LED-Scoreboard/mlb-led-scoreboard #574: Scoreboard crashing after displaying 2 of the upcoming games for today

ando01, 2025-02-21:
I have updated to the latest release (8.10) and started my scoreboard today to get some info on the ST games coming up today. After displaying only 2 of the 6 games on the schedule, the scoreboard crashes and reboots. Below is the error that shows at the time of the crash:

After some research it seems the crash is being caused by a KeyError when trying to access a team's abbreviation using an ID that doesn't exist in the teams.TEAM_ID_ABBR dictionary. The specific ID causing the issue is 343. I cannot seem to find an ID 343 so I am posting here to see if someone can help.

Thanks

`ERROR (07:50:14): Untrapped error in main!
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]: Traceback (most recent call last):
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:   File "/home/andy/mlb-led-scoreboard/main.py", line 169, in <module>
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:     main(matrix, config)
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:   File "/home/andy/mlb-led-scoreboard/main.py", line 86, in main
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:     __refresh_gameday(render, data)
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:   File "/home/andy/mlb-led-scoreboard/main.py", line 153, in __refresh_gameday
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:     data.advance_to_next_game()
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:   File "/home/andy/mlb-led-scoreboard/data/__init__.py", line 86, in advance_to_next_game
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:     self.print_game_data_debug()
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:   File "/home/andy/mlb-led-scoreboard/data/__init__.py", line 146, in print_game_data_debug
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:     debug.log("Pre: %s", Pregame(self.current_game, self.config.time_format))
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:   File "/home/andy/mlb-led-scoreboard/data/scoreboard/pregame.py", line 12, in __init__
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:     self.away_team = game.away_abbreviation()
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:                      ^^^^^^^^^^^^^^^^^^^^^^^^
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:   File "/home/andy/mlb-led-scoreboard/data/game.py", line 118, in away_abbreviation
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:     return teams.TEAM_ID_ABBR[self._current_data["gameData"]["teams"]["away"]["id"]]
Feb 21 07:50:14 pi64x64MLBScoreboard.local main.py[2955]:            ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^`

WardBrian, 2025-02-21:
Oh, I bet this is the Huskies (Boston always plays Northeastern University during spring training)

Ill have to think about the best way around this

ando01, 2025-02-21:
Ah I bet you are right. That the game I want to watch too :-)

WardBrian, 2025-02-21:
A quick fix is edit data/teams.py to have a mapping for that ID. We will have to make it more robust, but probably not in time for the game today!

ando01, 2025-02-21:
Yep that worked! Here is the updates teams.py

`# the following were created like so
# import statsapi
# _teams = statsapi.get('teams', {'sportIds':1})['teams']
# TEAM_ID_ABBR = {t["id"]: t["abbreviation"] for t in _teams}
# TEAM_ID_NAME = {t["id"]: t["teamName"] for t in _teams}
# TEAM_NAME_ID = {t["teamName"]: t["id"] for t in _teams}

# Can be customized, but will require changing in colors/teams.json as well
TEAM_ID_ABBR = {
    133: "ATH",
    134: "PIT",
    135: "SD",
    136: "SEA",
    137: "SF",
    138: "STL",
    139: "TB",
    140: "TEX",
    141: "TOR",
    142: "MIN",
    143: "PHI",
    144: "ATL",
    145: "CWS",
    146: "MIA",
    147: "NYY",
    158: "MIL",
    108: "LAA",
    109: "AZ",
    110: "BAL",
    111: "BOS",
    112: "CHC",
    113: "CIN",
    114: "CLE",
    115: "COL",
    116: "DET",
    117: "HOU",
    118: "KC",
    119: "LAD",
    120: "WSH",
    121: "NYM",
    343: "NEU",
}

# Can be customized
TEAM_ID_NAME = {
    133: "Athletics",
    134: "Pirates",
    135: "Padres",
    136: "Mariners",
    137: "Giants",
    138: "Cardinals",
    139: "Rays",
    140: "Rangers",
    141: "Blue Jays",
    142: "Twins",
    143: "Phillies",
    144: "Braves",
    145: "White Sox",
    146: "Marlins",
    147: "Yankees",
    158: "Brewers",
    108: "Angels",
    109: "D-backs",
    110: "Orioles",
    111: "Red Sox",
    112: "Cubs",
    113: "Reds",
    114: "Guardians",
    115: "Rockies",
    116: "Tigers",
    117: "Astros",
    118: "Royals",
    119: "Dodgers",
    120: "Nationals",
    121: "Mets",
    343: "Northeastern Univerisity"
}

# Can be customized, but names in the config.json file must match
TEAM_NAME_ID = {
    "Athletics": 133,
    "Pirates": 134,
    "Padres": 135,
    "Mariners": 136,
    "Giants": 137,
    "Cardinals": 138,
    "Rays": 139,
    "Rangers": 140,
    "Blue Jays": 141,
    "Twins": 142,
    "Phillies": 143,
    "Braves": 144,
    "White Sox": 145,
    "Marlins": 146,
    "Yankees": 147,
    "Brewers": 158,
    "Angels": 108,
    "D-backs": 109,
    "Orioles": 110,
    "Red Sox": 111,
    "Cubs": 112,
    "Reds": 113,
    "Guardians": 114,
    "Rockies": 115,
    "Tigers": 116,
    "Astros": 117,
    "Royals": 118,
    "Dodgers": 119,
    "Nationals": 120,
    "Mets": 121,
    "Northesatern": 343,
}
`

ty-porter, 2025-02-22:
Released fix in #575 as part of #569.

PR #575 (2025-02-22): Fall back to API data when team ID not recognized 
Fixes #574

This still allows customization in teams.py, but uses the same data as before #556 if the id isn't found, like is common in exhibition games
closed 2025-02-22 by ty-porter