MLB-LED-Scoreboard/mlb-led-scoreboard #628: Add configurable Header Values to API requests

tracerrx, 2025-06-26:
Looking at the `statsapi.get()` it appears that brian has already done the work and there is a `testing_params` already integrated.. We could just set that equal to a config.json var  like "additional_api_parameters"

some more info for later:
https://github.com/toddrob99/MLB-StatsAPI/wiki/Function:-get

`request_kwargs` is a dict of additional parameters to pass to the underlying `requests.get call`. Of particular interest is `request_kwargs={"headers":{"Accept-Encoding": "gzip"}}, `which is recommended by MLB for some larger endpoints like game. Defaults to {}

Another Example:
```
# Define the custom headers you want to send as a dictionary
custom_headers = {
    'Accept-Encoding': 'gzip',  # Example: requesting gzipped response
    'User-Agent': 'My Awesome MLB Stats API Script/1.0',  # Example: identifying your application
    # Add any other headers as key-value pairs
}

# Use statsapi.get() to retrieve game information, passing the headers in request_kwargs
game_data = statsapi.get('game', {'gamePk': game_id}, request_kwargs={'headers': custom_headers})
```

WardBrian, 2025-06-26:
I think we could probably be passing the gzip header by default, I am just a little worried about the extra compute overhead for low powered boards

tracerrx, 2025-06-26:
Agreed, Ty was asking for configurable headers in the discord... I'm not sure what his plans for them are (maybe setting the user-agent to **YANKEES 4EVER**).  I noticed the gzip header in the statsapi documentation and figured I would paste it hear so we could discuss.  Maybe we can inject the gzip header based if the pi is greater than a 2? `os.path.exists(model_path)` Not sure this works across different rpi's os though.

ty-porter, 2025-06-27:
With #626 on the horizon I was considering adding user-agent to request headers on Stats API calls as a best practice. With our user base and 10s refresh duration we probably contribute a decent amount of traffic. Some users could 3x that rate

tracerrx, 2025-06-27:
Pretty sure I have this working for all the statsapi.game() requests in dates, game, standings, team, unifiorms..... No live game on currently but not getting any errors and it is loading the schedule and standings.  

What do we think is the simplest way to see if my headers are actually being sent and formatted correctly or can I assume it's working if the API requests aren't failing??

WardBrian, 2025-06-27:
Simplest way is probably enabling verbose logging, which will show the requests:

Basically, if you drop this somewhere in main:
```
from http.client import HTTPConnection
HTTPConnection.debuglevel = 1
import logging
logging.basicConfig(level=logging.DEBUG)
```
all of our api requests should log a bunch of information, including their headers

tracerrx, 2025-06-27:
That works great thanks!  

On the positive, it appears the gzip headers are being sent by default, but for the life of me I can't figure out how to override the headers... 

```
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): statsapi.mlb.com:443
send: b'GET /api/v1/uniforms/game?gamePks=777332&fields=uniforms,home,away,uniformAssets,uniformAssetText HTTP/1.1\r\nHost: statsapi.mlb.com\r\nUser-Agent: python-requests/2.32.3\r\nAccept-Encoding: gzip, deflate\r\nAccept: */*\r\nConnection: keep-alive\r\n\r\n'
```

tracerrx, 2025-06-27:
I think I have it... whew... just have to refactor some stuff

ty-porter, 2025-07-10:
Closing this, we implemented this in 8.5.0